oxidecomputer/hubris #2031: `humility dump` shows `DumpAreaInUse`

mkeeter, 2025-03-17:
While debugging https://github.com/oxidecomputer/colo/issues/108, we _also_ saw an issue with Hubris not being able to take a dump:
```
augustus@mgmt:~/mupdate_2025-03-06$ humility --ip fe80::aa40:25ff:fe04:24d%enp9s0f0.401 --archive repo/targets/c7fdab7f3f36f60c2aa433b9383a5e1df723f605f4a7d09c17662a9de4e39b41.gimlet_sp-gimlet-c-1.0.34.tar.gz dump
humility: connecting to fe80::aa40:25ff:fe04:24d%9
humility: using UDP dump agent
humility: initializing dump agent state
humility: initializing segments
humility dump failed: invalid response: Err(DumpAreaInUse)
```

I wasn't in the meeting at this point, but first-hand reports indicate that `humility dump -l` did not show any dumps taking up space, and running a different `humility dump` subcommand accidentally cleared dump space before we could figure out the root cause.  We should watch the recording to figure out exactly what steps were taken.

cbiffle, 2025-03-17:
I am concerned that our current dump area access control model may effectively prevent readers of the dump area; ideally we'd be able to read while someone is writing, so that a crash-looping task won't block us out.

I bet we haven't seen this in practice because crash-looping tasks quickly _fill the entire dump area._

That said, it sounds like the sled in question saw no crashes, so this is probably not the problem right now.

mkeeter, 2025-03-17:
...and more weirdness in the `jefe` logs (`/staff/rack3/BRM42220001/hubris.core.0`):
```
matt@jeeves /staff/rack3/BRM42220001 () $ h ringbuf jefe
humility: attached to dump
humility: ring buffer task_jefe::dump::__RINGBUF in jefe:
 NDX LINE      GEN    COUNT PAYLOAD
   7  100       20        1 GetDumpArea(0x1)
   0  101       21        1 Base(0x3800e000)
   1  100       21        1 GetDumpArea(0x1)
   2  101       21        1 Base(0x3800f000)
   3  110       21        1 GetDumpAreaFailed(InvalidIndex)
   4  100       21        1 GetDumpArea(0x0)
   5  101       21        1 Base(0x30010000)
   6   73       21        1 Initialized
```

Looking at the ringbufs for `dump-agent`, it looks like it only calls `SpRotDump` once:
```
matt@jeeves /staff/rack3/BRM42220001 () $ h ringbuf dump_agent
humility: attached to dump
humility: ring buffer task_dump_agent::__RINGBUF in dump_agent:
 NDX LINE      GEN    COUNT PAYLOAD
   0  159        1        1 SpRotDump
```

...so this implies that the earlier failure is in loading the dump segments (?).

mkeeter, 2025-03-17:
Reading through more code, this part represents incremental claiming of each dump region:
```
 NDX LINE      GEN    COUNT PAYLOAD
   7  100       20        1 GetDumpArea(0x1)
   0  101       21        1 Base(0x3800e000)
   1  100       21        1 GetDumpArea(0x1)
   2  101       21        1 Base(0x3800f000)
   3  110       21        1 GetDumpAreaFailed(InvalidIndex)
```

Here are a few more relevant counters from `jefe`:

```
dump_agent
 |
 +---> drv_sprot_api::__SPROT_CLIENT_COUNTERS:
 |     <no counts recorded>
 +---> task_jefe_api::__JEFE_CLIENT_COUNTERS:
 |       118 get_dump_area(Ok)
 |         1 get_dump_area(Err(InvalidArea))
 |         3 claim_dump_area(Ok)
 |         3 claim_dump_area(Err(DumpAreaInUse))
 |         5 dump_task_region(Err(DumpAreaInUse))
 |         4 reinitialize_dump_areas(Ok)
 +---> task_net_api::__NET_CLIENT_COUNTERS:
    247859989 recv_packet(Err(QueueEmpty))
         142 recv_packet(Ok)
         141 send_packet(Ok)
```

I think many of the failures are interactions from the operator, e.g. trying to do `humility host` command (which read data using `dump_task_region`).

mkeeter, 2025-03-19:
We encountered this again when debugging a rack at Benchmark.

Doing `humility dump --initialize-dump-agent` allows us to do single-task dumps, which look like what you'd expect:
```
BRM27230047-switch # humility dump --dump-agent-status | head -n100
humility: connecting to fe80::aa40:25ff:fe04:786%50
humility: using UDP dump agent
[
    (
        DumpAreaHeader {
            magic: [
                0x1,
                0xde,
                0xde,
                0xad,
            ],
            address: 0x30010000,
            length: 0x1000,
            written: 0xfdc,
            contents: 0x1,
            dumper: 0x2,
            nsegments: 0x2,
            next: 0x30011000,
        },
        Some(
            DumpTask {
                magic: [
                    0xda,
                    0xda,
                ],
                id: 0x13,
                pad: 0x0,
                time: 0x392cfb2,
            },
        ),
    ),
    (
        DumpAreaHeader {
            magic: [
                0x1,
                0xde,
                0xde,
                0xad,
            ],
            address: 0x30011000,
            length: 0x1000,
            written: 0xf3c,
            contents: 0x1,
            dumper: 0x2,
            nsegments: 0x0,
            next: 0x30012000,
        },
        None,
    ),
    (
        DumpAreaHeader {
            magic: [
                0x1,
                0xde,
                0xde,
                0xad,
            ],
            address: 0x30012000,
            length: 0x1000,
            written: 0x430,
            contents: 0x1,
            dumper: 0x2,
            nsegments: 0x0,
            next: 0x30013000,
        },
        None,
    ),
    (
        DumpAreaHeader {
            magic: [
                0x1,
                0xde,
                0xde,
                0xad,
            ],
            address: 0x30013000,
            length: 0x1000,
            written: 0x18,
            contents: 0x0,
            dumper: 0xff,
            nsegments: 0x0,
            next: 0x30014000,
        },
        None,
    ),
    (
        DumpAreaHeader {
            magic: [
                0x1,
                0xde,
                0xde,
                0xad,
            ],
            address: 0x30014000,
            length: 0x1000,
            written: 0x18,
            contents: 0x0,
            dumper: 0xff,
            nsegments: 0x0,
            next: 0x30015000,
        },
        None,
```

mkeeter, 2025-03-19:
This was an issue with `DumpAreaHeader` caching:

- `Jefe::dump_area` returned a stale `DumpAreaHeader` that falsely reported the dump area as unclaimed
- Then `dump_agent` tried to claim it
- This failed because it was, in fact, claimed.

PR #2032 (2025-03-19): Fix dump area caching
The address in the dump area header is constant, but other values can change, so we should reload it from RAM!

Fixes #2031
commit aaf4136b (2025-03-21): Fix dump area caching (#2032)

The address in the dump area header is constant, but other values can
change, so we should reload it from RAM!

Fixes #2031