oxidecomputer/hubris #2056: `app/lpc55xpresso/app-sprot.toml` is busted on `master`

hawkw, 2025-05-02:
Discovered while going through all the `app.toml`s and testing if they build with the `zerocopy` update as part of #2054, it looks like this TOML doesn't build (and isn't checked on CI). It seems the SWD driver is missing some notifications, I _assume_ adding them will fix it:
```
error: could not compile `drv-lpc55-swd` (bin "drv-lpc55-swd") due to 27 previous errors
Error: failed to build swd

Caused by:
    Missing notifications; do you need to add them to your TOML file?
    - swd is missing ["jtag-detect-irq", "sp-reset-irq", "timer"]
```

<details>
<summary>Full build output (lengthy(</summary>

```console
eliza@theseus ~/Code/oxide/hubris $ cargo xtask dist app/lpc55xpresso/app-sprot.toml
    Finished `dev` profile [optimized + debuginfo] target(s) in 0.22s
     Running `target/debug/xtask dist app/lpc55xpresso/app-sprot.toml`
app.toml has changed; rebuilding all tasks
cleaning ["lpc55xpresso", "task-jefe", "task-hiffy", "task-idle", "lpc55-update-server", "drv-lpc55-syscon", "drv-lpc55-gpio", "drv-user-leds", "drv-lpc55-usart", "drv-lpc55-i2c", "drv-lpc55-rng", "task-pong", "drv-lpc55-swd", "task-dumper", "drv-lpc55-sprot-server", "task-attest"]
     Removed 223 files, 509.4MiB total
building crate task-jefe
   Compiling task-jefe v0.1.0 (/home/eliza/Code/oxide/hubris/task/jefe)
   Compiling hubris-num-tasks v0.1.0 (/home/eliza/Code/oxide/hubris/sys/num-tasks)
    Finished `release` profile [optimized + debuginfo] target(s) in 0.97s
target/thumbv8m.main-none-eabihf/release/task-jefe -> target/lpc55xpresso-sprot/dist/jefe.elf
building crate task-hiffy
   Compiling drv-lpc55-gpio-api v0.1.0 (/home/eliza/Code/oxide/hubris/drv/lpc55-gpio-api)
   Compiling hubris-num-tasks v0.1.0 (/home/eliza/Code/oxide/hubris/sys/num-tasks)
   Compiling task-hiffy v0.1.0 (/home/eliza/Code/oxide/hubris/task/hiffy)
    Finished `release` profile [optimized + debuginfo] target(s) in 0.84s
target/thumbv8m.main-none-eabihf/release/task-hiffy -> target/lpc55xpresso-sprot/dist/hiffy.elf
building crate task-idle
   Compiling task-idle v0.1.0 (/home/eliza/Code/oxide/hubris/task/idle)
    Finished `release` profile [optimized + debuginfo] target(s) in 0.27s
target/thumbv8m.main-none-eabihf/release/task-idle -> target/lpc55xpresso-sprot/dist/idle.elf
building crate lpc55-update-server
   Compiling lpc55-update-server v0.1.0 (/home/eliza/Code/oxide/hubris/drv/lpc55-update-server)
   Compiling drv-lpc55-syscon-api v0.1.0 (/home/eliza/Code/oxide/hubris/drv/lpc55-syscon-api)
    Finished `release` profile [optimized + debuginfo] target(s) in 1.56s
target/thumbv8m.main-none-eabihf/release/lpc55-update-server -> target/lpc55xpresso-sprot/dist/update_server.elf
building crate drv-lpc55-syscon
   Compiling drv-lpc55-syscon v0.1.0 (/home/eliza/Code/oxide/hubris/drv/lpc55-syscon)
   Compiling drv-lpc55-syscon-api v0.1.0 (/home/eliza/Code/oxide/hubris/drv/lpc55-syscon-api)
    Finished `release` profile [optimized + debuginfo] target(s) in 0.67s
target/thumbv8m.main-none-eabihf/release/drv-lpc55-syscon -> target/lpc55xpresso-sprot/dist/syscon_driver.elf
building crate drv-lpc55-gpio
   Compiling drv-lpc55-gpio v0.1.0 (/home/eliza/Code/oxide/hubris/drv/lpc55-gpio)
   Compiling drv-lpc55-gpio-api v0.1.0 (/home/eliza/Code/oxide/hubris/drv/lpc55-gpio-api)
    Finished `release` profile [optimized + debuginfo] target(s) in 0.78s
target/thumbv8m.main-none-eabihf/release/drv-lpc55-gpio -> target/lpc55xpresso-sprot/dist/gpio_driver.elf
building crate drv-user-leds
   Compiling enum-map v2.7.3
   Compiling drv-user-leds v0.1.0 (/home/eliza/Code/oxide/hubris/drv/user-leds)
   Compiling drv-user-leds-api v0.1.0 (/home/eliza/Code/oxide/hubris/drv/user-leds-api)
   Compiling drv-lpc55-gpio-api v0.1.0 (/home/eliza/Code/oxide/hubris/drv/lpc55-gpio-api)
cargo:rerun-if-env-changed=HUBRIS_TASK_CONFIG
cargo:rerun-if-env-changed=HUBRIS_TASK_NAME
    Finished `release` profile [optimized + debuginfo] target(s) in 0.72s
target/thumbv8m.main-none-eabihf/release/drv-user-leds -> target/lpc55xpresso-sprot/dist/user_leds.elf
building crate drv-lpc55-usart
   Compiling build-lpc55pins v0.1.0 (/home/eliza/Code/oxide/hubris/build/lpc55pins)
   Compiling drv-lpc55-usart v0.1.0 (/home/eliza/Code/oxide/hubris/drv/lpc55-usart)
    Finished `release` profile [optimized + debuginfo] target(s) in 0.96s
target/thumbv8m.main-none-eabihf/release/drv-lpc55-usart -> target/lpc55xpresso-sprot/dist/usart_driver.elf
building crate drv-lpc55-i2c
   Compiling drv-lpc55-i2c v0.1.0 (/home/eliza/Code/oxide/hubris/drv/lpc55-i2c)
    Finished `release` profile [optimized + debuginfo] target(s) in 0.50s
target/thumbv8m.main-none-eabihf/release/drv-lpc55-i2c -> target/lpc55xpresso-sprot/dist/i2c_driver.elf
building crate drv-lpc55-rng
   Compiling num-traits v0.2.19
[... 242 lines cut ...]
1974 |     sys_irq_control(notifications::JTAG_DETECT_IRQ_MASK, true);
     |                                    ^^^^^^^^^^^^^^^^^^^^ not found in `notifications`

error[E0425]: cannot find function `setup_rot_to_sp_reset_l_out` in this scope
    --> drv/lpc55-swd/src/main.rs:1366:9
     |
1366 |         setup_rot_to_sp_reset_l_out(self.gpio);
     |         ^^^^^^^^^^^^^^^^^^^^^^^^^^^ not found in this scope

error[E0425]: cannot find function `setup_rot_to_sp_reset_l_in` in this scope
    --> drv/lpc55-swd/src/main.rs:1373:9
     |
1373 |         setup_rot_to_sp_reset_l_in(self.gpio);
     |         ^^^^^^^^^^^^^^^^^^^^^^^^^^ not found in this scope

For more information about this error, try `rustc --explain E0425`.
error: could not compile `drv-lpc55-swd` (bin "drv-lpc55-swd") due to 27 previous errors
Error: failed to build swd

Caused by:
    Missing notifications; do you need to add them to your TOML file?
    - swd is missing ["jtag-detect-irq", "sp-reset-irq", "timer"]
```

</details>

aapoalas, 2025-05-03:
https://github.com/oxidecomputer/hubris/pull/1920 PR of mine fixed it at least when I created it; probably it would fix it today as well.

labbott, 2025-05-05:
I'd vote for this app configuration to be deleted per the top

```
# This is a variation on the main LPC55xpresso `app.toml` that allows it to be
# used as an SP-RoT tester, attached to a Gimletlet.  Only the SPI target should
# work; using the LPC55Xpresso as a debugger for the Gimletlet would require
# additional changes to this file.
```

We have many many better options at this point

labbott, 2025-06-16:
This is removed

closed 2025-06-16 by labbott