oxidecomputer/hubris #2070: `transceivers` / QSFP FPGA hang on Berlin Sidecar

mkeeter, 2025-05-16:
In the Berlin environment, we noticed that the host couldn't get MAC addresses from the SP on `switch1` (`BRM42220082` is the relevant scrimlet).

After a few false leads, it turned out that we could talk to `control-plane-agent` fine, but `transceivers` (which reports MAC addresses to the host) was stuck.

Sampling a few times, we appear to be stuck in [this loop](https://github.com/oxidecomputer/hubris/blob/7a24b884ba1907c1d45432760ad189d1a606e4a2/drv/sidecar-front-io/src/transceivers.rs#L1054-L1075):
```
root@oxz_switch1:~# /tmp/humility-0.12.0-probless --archive /tmp/21f65cb23eaacf2a847f905cd443daf05d85ef6717a255ef28ae0cef23a99432.switch_sp-sidecar-c-1.0.37.tar.gz --ip fe80::aa40:25ff:fe05:1f00%sidecar0 tasks -sl transceivers
humility: connecting to fe80::aa40:25ff:fe05:1f00%38
humility: reading tasks remotely; state may not be consistent
system time = unavailable-via-net
ID TASK                       GEN PRI STATE
16 transceivers                 0   6 notif: bit31(T=93631248)
   |
   +--->  0x2404c7c8 0x0807f9dc userlib::sys_recv_stub
                     @ /hubris/sys/userlib/src/lib.rs:368
          0x2404c810 0x0807fa64 userlib::sys_get_timer
                     @ /hubris/sys/userlib/src/lib.rs:1130
          0x2404c810 0x0807fa3c userlib::hl::sleep_until
                     @ /hubris/sys/userlib/src/hl.rs:425
          0x2404c810 0x0807fa64 userlib::hl::sleep_for
                     @ /hubris/sys/userlib/src/hl.rs:459
          0x2404c840 0x080784b2 drv_sidecar_front_io::transceivers::Transceivers::get_i2c_status_and_read_buffer
          0x2404d000 0x0807c4f8 <core::result::Result<T,E> as core::ops::try_trait::Try>::branch
                     @ /rustc/c52c23b6f44cd19718721a5e3b2eeb169e9c96ff/library/core/src/result.rs:1985
          0x2404d000 0x0807c4f8 main
                     @ /hubris/drv/transceivers-server/src/main.rs:619

root@oxz_switch1:~# /tmp/humility-0.12.0-probless --archive /tmp/21f65cb23eaacf2a847f905cd443daf05d85ef6717a255ef28ae0cef23a99432.switch_sp-sidecar-c-1.0.37.tar.gz --ip fe80::aa40:25ff:fe05:1f00%sidecar0 tasks -sl transceivers
humility: connecting to fe80::aa40:25ff:fe05:1f00%38
humility: reading tasks remotely; state may not be consistent
system time = unavailable-via-net
ID TASK                       GEN PRI STATE
16 transceivers                 0   6 notif: bit31(T=93710054)
   |
   +--->  0x2404c7c8 0x0807f9dc userlib::sys_recv_stub
                     @ /hubris/sys/userlib/src/lib.rs:368
          0x2404c810 0x0807fa64 userlib::sys_get_timer
                     @ /hubris/sys/userlib/src/lib.rs:1130
          0x2404c810 0x0807fa3c userlib::hl::sleep_until
                     @ /hubris/sys/userlib/src/hl.rs:425
          0x2404c810 0x0807fa64 userlib::hl::sleep_for
                     @ /hubris/sys/userlib/src/hl.rs:459
          0x2404c840 0x080784b2 drv_sidecar_front_io::transceivers::Transceivers::get_i2c_status_and_read_buffer
          0x2404d000 0x0807c4f8 <core::result::Result<T,E> as core::ops::try_trait::Try>::branch
                     @ /rustc/c52c23b6f44cd19718721a5e3b2eeb169e9c96ff/library/core/src/result.rs:1985
          0x2404d000 0x0807c4f8 main
                     @ /hubris/drv/transceivers-server/src/main.rs:619
```

Looking at the FPGA task, we're repeating the same SPI message over and over:
```
root@oxz_switch1:~# /tmp/humility-0.12.0-probless --archive /tmp/21f65cb23eaacf2a847f905cd443daf05d85ef6717a255ef28ae0cef23a99432.switch_sp-sidecar-c-1.0.37.tar.gz --ip fe80::aa40:25ff:fe05:1f00%sidecar0 ringbuf ecp5_front_io
humility: connecting to fe80::aa40:25ff:fe05:1f00%38
humility: ring buffer drv_fpga_devices::ecp5::__RINGBUF in ecp5_front_io:
humility: ring buffer drv_fpga_server::__RINGBUF in ecp5_front_io:
humility: ring buffer drv_stm32h7_spi_server_core::__RINGBUF in ecp5_front_io:
   TOTAL VARIANT
283276834 WaitISR
191326918 Tx
191326918 Rx
47831753 Start(write)
46054995 Start(read)
 NDX LINE      GEN    COUNT PAYLOAD
  37  469    55120        1 Tx(0x0)
  38  533    55120        2 WaitISR(0x10012)
  39  503    55120        1 Rx(0x30)
  40  359    55120        1 Start(write, (0x3, 0x0))
  41  469    55120        1 Tx(0x1)
  42  469    55120        1 Tx(0x2)
  43  469    55120        1 Tx(0x8)
  44  533    55120        2 WaitISR(0x30012)
  45  503    55120        1 Rx(0x0)
  46  533    55120        1 WaitISR(0x20012)
  47  503    55120        1 Rx(0x0)
  48  533    55120        1 WaitISR(0x1301a)
  49  503    55120        1 Rx(0x0)
  50  359    55120        1 Start(read, (0x0, 0x1))
  51  469    55120        1 Tx(0x0)
  52  533    55120        2 WaitISR(0x10012)
  53  503    55120        1 Rx(0x30)
  54  359    55120        1 Start(write, (0x3, 0x0))
  55  469    55120        1 Tx(0x1)
  56  469    55120        1 Tx(0x2)
  57  469    55120        1 Tx(0x8)
  58  533    55120        2 WaitISR(0x30012)
  59  503    55120        2 Rx(0x0)
  60  533    55120        1 WaitISR(0x10012)
  61  503    55120        1 Rx(0x0)
  62  359    55120        1 Start(read, (0x0, 0x1))
  63  469    55120        1 Tx(0x0)
   0  533    55121        2 WaitISR(0x10012)
   1  503    55121        1 Rx(0x30)
   2  359    55121        1 Start(write, (0x3, 0x0))
   3  469    55121        1 Tx(0x1)
   4  469    55121        1 Tx(0x2)
   5  469    55121        1 Tx(0x8)
   6  533    55121        1 WaitISR(0x30012)
   7  533    55121        1 WaitISR(0x22017)
   8  503    55121        3 Rx(0x0)
   9  359    55121        1 Start(read, (0x0, 0x1))
  10  469    55121        1 Tx(0x0)
  11  533    55121        2 WaitISR(0x10012)
  12  503    55121        1 Rx(0x30)
  13  359    55121        1 Start(write, (0x3, 0x0))
  14  469    55121        1 Tx(0x1)
  15  469    55121        1 Tx(0x2)
  16  469    55121        1 Tx(0x8)
  17  533    55121        2 WaitISR(0x30012)
  18  503    55121        1 Rx(0x0)
  19  533    55121        1 WaitISR(0x22017)
  20  503    55121        2 Rx(0x0)
  21  359    55121        1 Start(read, (0x0, 0x1))
  22  469    55121        1 Tx(0x0)
  23  533    55121        2 WaitISR(0x10012)
  24  503    55121        1 Rx(0x30)
  25  359    55121        1 Start(write, (0x3, 0x0))
  26  469    55121        1 Tx(0x1)
  27  469    55121        1 Tx(0x2)
  28  469    55121        1 Tx(0x8)
  29  533    55121        2 WaitISR(0x30012)
  30  503    55121        2 Rx(0x0)
  31  533    55121        1 WaitISR(0x10012)
  32  503    55121        1 Rx(0x0)
  33  359    55121        1 Start(read, (0x0, 0x1))
  34  469    55121        1 Tx(0x0)
  35  533    55121        1 WaitISR(0x10012)
  36  503    55121        1 Rx(0x30)
```

We're sending `[0x01, 0x02, 0x08]` and receiving `0x30`.  The transmitted message is decoded as `[ReadOp::Read, QSFP_PORT0_STATUS (0x208)]`, and the returned byte is `BUSY | WDATA_FIFO_EMPTY`.

Hubris could be more robust here, but it seems like the FPGA has gotten stuck, which isn't great.  The hold on Berlin was extended through Monday, so it should still be like this when @Aaron-Hartwig gets back.

Aaron-Hartwig, 2025-05-19:
I agree with Matt's assessment here. It's not clear to me what happened, but my assumption is that we somehow got a bus glitch that threw the FPGA state machine off. The infinite looping is expected since the FPGA doesn't have a per-transaction timeout, and the SP assumes every transaction it starts will eventually finish (hence the looping waiting for the `BUSY` bit to clear). Given the rarity of this event (we've never seen this before?), I think it makes the most sense to focus on recovering gracefully when something went awry.

I'd propose the following:
- FPGA
  - Create a timeout value for each I2C bus
    - 10 ms is ~100x the max transaction time? (100KHz = 10us bit period, so 10us * 9 bits/byte * 128 bytes = 11520us)
    - If the transaction times out, tell our I2C peripheral to abort its transaction (clearing the `BUSY` bit) and return an error (I'd extend the current `FpgaI2CFailure` enum with a new variant)
- SP
  - No change! In the [loop linked above](https://github.com/oxidecomputer/hubris/blob/7a24b884ba1907c1d45432760ad189d1a606e4a2/drv/sidecar-front-io/src/transceivers.rs#L1054-L1075) it would now have an exit path since the new FPGA would have a timeout. Should that occur, `status.done` would be true and `status.error` would contain an error saying what happened.

mkeeter, 2025-11-25:
We believe we've seen this again today in a customer update.  We saw the same behavior of being able to talk to `control-plane-agent` but not `transceivers`, and the host not being able to get MAC addresses.

The dump is at `/staff/*/mupdate-20251125-v17.1/hubris.core.0` (customer name redacted, but it's a unique wildcard path)

The transceivers task is spinning in the same place:
```
system time = 7951398
ID TASK                       GEN PRI STATE
16 transceivers                 0   6 notif: bit31(T+1)
   |
   +--->  0x240505b0 0x080068fc userlib::sys_recv_stub
          0x24050618 0x0800dd25 userlib::sys_get_timer
                     @ /hubris/sys/userlib/src/lib.rs:1172:9
          0x24050618 0x0800dd25 userlib::hl::sleep_until
                     @ /hubris/sys/userlib/src/hl.rs:440:12
          0x24050618 0x0800dd25 userlib::hl::sleep_for
                     @ /hubris/sys/userlib/src/hl.rs:471:5
          0x24050840 0x080094c2 drv_sidecar_front_io::transceivers::Transceivers::get_i2c_status_and_read_buffer
                     @ /hubris/drv/sidecar-front-io/src/transceivers.rs:1060:9
          0x24050840 0x080094c2 drv_transceivers_server::udp::<impl drv_transceivers_server::ServerImpl>::read
                     @ /hubris/drv/transceivers-server/src/udp.rs:1169:18
          0x24050840 0x080094c2 drv_transceivers_server::udp::<impl drv_transceivers_server::ServerImpl>::handle_host_request
                     @ /hubris/drv/transceivers-server/src/udp.rs:394:38
          0x24050840 0x080094c2 drv_transceivers_server::udp::<impl drv_transceivers_server::ServerImpl>::handle_message
                     @ /hubris/drv/transceivers-server/src/udp.rs:254:49
          0x24050840 0x080094c2 drv_transceivers_server::udp::<impl drv_transceivers_server::ServerImpl>::handle_packet
                     @ /hubris/drv/transceivers-server/src/udp.rs:115:34
          0x24051000 0x0800b668 drv_transceivers_server::udp::<impl drv_transceivers_server::ServerImpl>::check_net
                     @ /hubris/drv/transceivers-server/src/udp.rs:92:30
          0x24051000 0x0800b668 main
                     @ /hubris/drv/transceivers-server/src/main.rs:758:14
```

There's also signs of receiving junk from the transceivers in their ringbuf:
```
humility: attached to dump
humility: ring buffer drv_transceivers_server::__RINGBUF in transceivers:
   TOTAL VARIANT
       5 UnknownInterface
       1 FrontIOBoardReady(true)
       1 LEDInit
       1 LEDInitComplete
       1 ModulePresenceUpdate
 NDX LINE      GEN    COUNT PAYLOAD
   0  696        1        1 FrontIOBoardReady(true)
   1  714        1        1 LEDInit
   2  172        1        1 LEDInitComplete
   3  559        1        1 ModulePresenceUpdate(LogicalPortMask(0x100))
   4  346        1        1 UnknownInterface(0x8, Unknown(0x49))
   5  346        1        1 UnknownInterface(0x8, Unknown(0x74))
   6  346        1        1 UnknownInterface(0x8, Unknown(0x6c))
   7  346        1        1 UnknownInterface(0x8, Unknown(0x43))
   8  346        1        1 UnknownInterface(0x8, Unknown(0x72))
humility: ring buffer drv_transceivers_server::udp::__RINGBUF in transceivers:
   TOTAL VARIANT
       1 Read
 NDX LINE      GEN    COUNT PAYLOAD
   0  369        1        1 Read(ModuleId(0x100), MemoryRead { page: Sff8636(Upper(UpperPage(0x0))), offset: 0x94, len: 0x48 })
```

`UnknownInterface` means that we got bytes that do not correspond to either CMIS (0x1E) or SFF8636 (0x0D or 0x11); this would be reasonable if the I2C interface was somehow corrupted / out of sync.

papertigers, 2026-08-25:
We saw this in berlin again while attempting to test a TUF repo of a draft omicron change.  @leftwo was in the process of mupdating the rack and we noticed things were an in an unexpected state.

```
mike@castle ~ $ pilot sp list
MAC               SERIAL      TYPE    IMAGE            IP
a8:40:25:04:01:80 BRM42220082 gimlet  35c296036a3feb4b fe80::aa40:25ff:fe04:180
a8:40:25:04:02:01 BRM42220011 gimlet  35c296036a3feb4b fe80::aa40:25ff:fe04:201
a8:40:25:04:04:82 BRM44220007 gimlet  35c296036a3feb4b fe80::aa40:25ff:fe04:482
a8:40:25:04:28:c7 271FVPY0    gimlet  35c296036a3feb4b fe80::aa40:25ff:fe04:28c7
a8:40:25:05:1f:00 BRM23230005 sidecar d4e8dc511b6f7221 fe80::aa40:25ff:fe05:1f00
a8:40:25:05:35:00 BRM23230003 sidecar d4e8dc511b6f7221 fe80::aa40:25ff:fe05:3500
a8:40:25:06:01:14 BRM11230009 psc     143aed42f6d335fa fe80::aa40:25ff:fe06:114
mike@castle ~ $ pilot tp login any
The illumos Project     helios-3.0.24216        August 2026
[null]root@oxz_switch0:~# pilot host list
CUBBY IP                        SERIAL      IMAGE
```

Alan noted that the mupdate left off at:
`[100%] Downloading installinator, waiting for it to start    `

Taking a look at installinators logs we see:

```
BRM44220007 # cat $(svcs -L installinator) | looker | more
[ Dec 28 00:00:02 Enabled. ]
[ Dec 28 00:00:42 Executing start method ("ctrun -l child -o noorphan,regent /op
t/oxide/installinator/installinator install --bootstrap-sled --from-ipcc --insta
ll-on-gimlet --stay-alive &"). ]
[ Dec 28 00:00:42 Method "start" exited with status 0. ]
Dec 28 00:00:42.979 INFO Starting mg-ddm service, file: installinator/src/bootst
rap.rs:50
Dec 28 00:00:42.980 INFO Importing mg-ddm service, path: /opt/oxide/mg-ddm/pkg/d
dm/manifest.xml, file: installinator/src/bootstrap.rs:129
Dec 28 00:00:43.167 INFO Setting mg-ddm interfaces, interfaces: ("cxgbe0/ll" "cx
gbe1/ll"), file: installinator/src/bootstrap.rs:139
Dec 28 00:00:43.176 INFO Enabling mg-ddm service, file: installinator/src/bootst
rap.rs:147
Dec 28 00:00:43.688 INFO Sending prefix to ddmd for advertisement, prefix: Ipv6N
et { addr: fdb0:a840:2504:492::, width: 64 }, DdmAdminClient: [::1]:8000, file:
installinator/src/bootstrap.rs:95
Dec 28 00:00:43.704 INFO discovered 0 peers: [], artifact: ArtifactLookupId { ki
nd: "installinator_document", hash: ArtifactHash(HexArray(a169490925e3adb5624fec
88c7843f7c9fd841afdd6f1871494ac175a8e689c0)) }, file: installinator/src/fetch.rs
:80
Dec 28 00:00:43.704 WARN unable to fetch artifact from peers, retrying discovery
, artifact: ArtifactLookupId { kind: "installinator_document", hash: ArtifactHas
h(HexArray(a169490925e3adb5624fec88c7843f7c9fd841afdd6f1871494ac175a8e689c0)) },
 file: installinator/src/fetch.rs:91
Dec 28 00:00:48.707 INFO discovered 0 peers: [], artifact: ArtifactLookupId { ki
nd: "installinator_document", hash: ArtifactHash(HexArray(a169490925e3adb5624fec
88c7843f7c9fd841afdd6f1871494ac175a8e689c0)) }, file: installinator/src/fetch.rs
:80
```

From the switch zone we started looking at some other services such as dendrite which was stuck trying to retrieve mac addresses:
```
11:41:26.351Z ERRO dpd: failed to send message within the retry limit
    limit = 3
    task = io
    unit = transceiver-controller
11:41:26.351Z ERRO dpd: failed to fetch MAC addresses from SP
    reason = MaxRetries(3)
11:41:27.656Z ERRO dpd: failed to send message within the retry limit
    limit = 3
    task = io
    unit = transceiver-controller
11:41:27.656Z ERRO dpd: failed to fetch MAC addresses from SP
    reason = MaxRetries(3)
```
[... 21 lines cut ...]
          0x240128f0 0x0809f181 userlib::hl::sleep_until
                     @ /hubris/sys/userlib/src/hl.rs:440:12
          0x240128f0 0x0809f181 userlib::hl::sleep_for
                     @ /hubris/sys/userlib/src/hl.rs:471:5
          0x24012920 0x08098418 <drv_front_io_api::transceivers::Transceivers>::get_i2c_status_and_read_buffer::<drv_front_io_api::transceivers::LogicalPort>
                     @ /hubris/drv/front-io-api/src/transceivers.rs:1061:9
          0x24013000 0x0809c082 <core::result::Result<drv_front_io_api::transceivers::PortI2CStatus, drv_fpga_api::FpgaError> as core::ops::try_trait::Try>::branch
                     @ /rustc/59807616e1fa2540724bfbac14d7976d7e4a3860/library/core/src/result.rs:2173:15
          0x24013000 0x0809c082 <drv_transceivers_server::ServerImpl>::get_transceiver_interface
                     @ /hubris/drv/transceivers-server/src/main.rs:304:22
          0x24013000 0x0809c082 <drv_transceivers_server::ServerImpl>::update_thermal_loop
                     @ /hubris/drv/transceivers-server/src/main.rs:367:28
          0x24013000 0x0809c082 <drv_transceivers_server::ServerImpl>::handle_spi_loop
                     @ /hubris/drv/transceivers-server/src/main.rs:564:14
          0x24013000 0x0809c082 main
                     @ /hubris/drv/transceivers-server/src/main.rs:749:32
```

We attempted to reset the sp, but it did not bring things back, which is what @Aaron-Hartwig expected would happen due to the i2c communication involving the fpga.  So, @jgallagher ran an ignition command to reset the sidecar which did bring things back into a working state. 

There is an sp dump here:
```
mike@castle ~ $ ls /staff/mike/berlin-hubris-2070
hubris.core.0
```

cbiffle, 2026-09-22:
So I notice that there's some code hidden inside the front-io-api crate that will happily spin _forever_ trying to talk to one transceiver. This is probably bad, in a task that's responsible for doing many different things.

https://github.com/oxidecomputer/hubris/blob/f6e5849734d4e7d965a2f1cd71ca4ed4b24532de/drv/front-io-api/src/transceivers.rs#L1082

This is currently where a machine is hung, and is also where @papertigers caught it hanging in the report above. This should probably at minimum have a retry count, but since we have many ports, we may want a state machine per port that we periodically revisit, instead of using a `loop`.

I hope that made sense, I just got here.

jamesmunns, 2026-09-22:
Nominating for R24 prio for me.