oxidecomputer/hubris #2072: Cosmo LM5066 says power is not good despite working fans

rmustacc, 2025-05-19:
With a Cosmo in A0 the `humility pmbus -s output` had the following:

```
$ pfexec humility pmbus -s
humility: attached to 0483:3754:003300324741500820383733 via ST-Link V3
DEVICE      RAIL               PG? #FLT       VIN      VOUT      IOUT    TEMP_1
tps546b24a  V3P3_SP_A2           Y    2   11.984V    3.309V    0.146A  40.750°C
tps546b24a  V5_SYS_A2            Y    2   11.984V    4.980V    1.945A  42.500°C
tps546b24a  V1P8_SYS_A2          Y    2   11.984V    1.791V   -0.566A  36.000°C
raa229620a  VDDCR_CPU0_A0        Y    0   12.070V    0.828V    1.900A  36.000°C
raa229620a  VDDCR_SOC_A0         Y    0   12.080V    0.874V   32.400A  43.000°C
raa229620a  VDDCR_CPU1_A0        Y    0   12.070V    0.810V    0.900A  39.000°C
raa229620a  VDDIO_SP5_A0         Y    0   12.070V    1.100V   17.800A  44.000°C
isl68224    V1P1_SP5_A0          Y    0   11.960V    1.100V    6.800A  39.000°C
isl68224    V1P8_SP5_A1          Y    0   11.950V    1.799V    3.800A  37.000°C
isl68224    V3P3_SP5_A1          Y    0   11.950V    2.749V    0.800A  38.000°C
tps546b24a  V0P96_NIC_VDD_A0HP   Y    0   11.984V    0.955V    5.766A  39.500°C
bmr491      V12_SYS_A2           Y    0   54.000V   12.001V    9.250A  38.250°C
lm5066      V54P5_FAN_EAST       N    3   54.742V   54.938V         -  36.250°C
lm5066      V54P5_FAN_CENTRAL    N    3   54.763V   55.025V         -  35.938°C
lm5066      V54P5_FAN_WEST       N    3   54.763V   55.003V         -  33.375°C
adm1272     V54P5_IBC_A3         Y    0    0x0892    0x0892    0x086d  34.976°C
```

Note how the LM5066 has PG set to false, despite the fact that the power is operating. So there is something that isn't quite right that we need to figure out here. I'm starting with the SP here given the evidence that it was working.

mkeeter, 2025-05-20:
More logs: Nathanael sees errors when reading the `lm5066`, and (more weirdly) the `raa229620a` parts are reporting voltages in hex (?!).

```
➜  hubris git:(master) humility pmbus -s
humility: attached via ST-Link V3
DEVICE      RAIL               PG? #FLT       VIN      VOUT      IOUT    TEMP_1
tps546b24a  V3P3_SP_A2           Y    2   12.000V    3.307V    0.322A  25.750°C
tps546b24a  V5_SYS_A2            Y    2   12.000V    4.988V    1.904A  28.750°C
tps546b24a  V1P8_SYS_A2          Y    2   11.984V    1.789V   -0.629A  25.000°C
raa229620a  VDDCR_CPU0_A0        Y    0    0x04ba    0x0380    0x005d    0x001e
raa229620a  VDDCR_SOC_A0         Y    0    0x04ba    0x0384    0x0092    0x001d
raa229620a  VDDCR_CPU1_A0        Y    0    0x04bc    0x037f    0x006c    0x001d
raa229620a  VDDIO_SP5_A0         Y    0    0x04bc    0x044b    0xfffd    0x001d
isl68224    V1P1_SP5_A0          Y    0   12.000V    1.099V    5.100A  28.000°C
isl68224    V1P8_SP5_A1          Y    0   12.010V    1.800V    1.200A  25.000°C
isl68224    V3P3_SP5_A1          Y    0   12.010V    2.749V    0.100A  25.000°C
tps546b24a  V0P96_NIC_VDD_A0HP   Y    0   11.984V    0.955V    1.934A  25.250°C
bmr491      V12_SYS_A2           Y    0   53.875V   11.985V    6.000A  31.250°C
lm5066      V54P5_FAN_EAST     --  error: can't read VOUT_MODE: NoRegister  --
lm5066      V54P5_FAN_CENTRAL  --  error: can't read VOUT_MODE: NoRegister  --
lm5066      V54P5_FAN_WEST     --  error: can't read VOUT_MODE: NoRegister  --
adm1272     V54P5_IBC_A3         Y    0    0x0892    0x0891    0x0847  25.929°C
```

mkeeter, 2025-05-20:
Nathanael's weirdness was an outdated Humility version; other issues should be addressed by #2073

closed 2025-06-09 by mkeeter