oxidecomputer/hubris #2075: LM5066 Current Scaling Incorrect

ericaasen, 2025-05-20:
On the LM5066 hot swap controllers on Cosmo, the current scaling is incorrect.  I suspect the values are being calculated based on a 5 milliohm current sense resistor, but this circuit uses a 7 milliohm sense resistor, so the readings are high by ~1.4x.

The "Reported Current" below is as reported by `humility sensors`

| Load Current | Reported Current |
|--------------|------------------|
| 1       | 1.4    |
| 2       | 2.82  |
| 3       | 4.24  |
| 3.5    | 4.94  |

There is also an ability to have the device average measurements for more accuracy/less noise:

[image]

There is a convenient way of getting all of the averaged telemetry information at once using a block read command as well:

[image]

I would set the averaging to no more than 256, which gives us an update rate of ~4Hz, which should be plenty fast for our purposes.

ericaasen, 2025-05-20:
full pmbus dump for one HSC with a 1A load:

```
λ humility pmbus -r V54P5_FAN_CENTRAL
humility: attached via ST-Link V3
0x01 OPERATION                 0x80
0x19 CAPABILITY                0xb0
0x43 VOUT_UV_WARN_LIMIT        0x0000 = 0.523V
0x4f OT_FAULT_LIMIT            0x0fff = 255.938°C
0x51 OT_WARN_LIMIT             0x0fff = 255.938°C
0x57 VIN_OV_WARN_LIMIT         0x0fff = 89.536V
0x58 VIN_UV_WARN_LIMIT         0x0000 = 0.262V
0x5d IIN_OC_WARN_LIMIT         0x0fff
0x78 STATUS_BYTE               0x03
0x79 STATUS_WORD               0x0203
0x7a STATUS_VOUT               0x00
0x7c STATUS_INPUT              0x00
0x7d STATUS_TEMPERATURE        0x00
0x7e STATUS_CML                0x82
0x7f STATUS_OTHER              0x20
0x80 STATUS_MFR_SPECIFIC       0x80
0x86 READ_EIN                  0x2f 0x4f 0xc4 0xb1 0x9f 0x3a           |/O...:
0x88 READ_VIN                  0x0795 = 42.577V
0x89 READ_IIN                  0x0429
0x8b READ_VOUT                 0x0794 = 42.817V
0x8d READ_TEMPERATURE_1        0x018a = 24.625°C
0x97 READ_PIN                  0x01f9
0x98 PMBUS_REVISION            0x22
0x99 MFR_ID                    0x54 0x49 0x00                          |TI.
0x9a MFR_MODEL                 0x4c 0x4d 0x35 0x30 0x36 0x36 0x49 0x00
0x9b MFR_REVISION              0x41 0x41                               |AA
0xd0 READ_VAUX                 0x0003 = 0.002V
0xd1 MFR_READ_IIN              0x042c
0xd2 MFR_READ_PIN              0x01fa
0xd3 MFR_IN_OC_WARN_LIMIT      0x0fff
0xd4 MFR_PIN_OP_WARN_LIMIT     0x0fff
0xd5 READ_PIN_PEAK             0x0700
0xd7 GATE_MASK                 0x00
0xd8 ALERT_MASK                0xfd04
0xd9 DEVICE_SETUP              0x00
0xda BLOCK_READ                0x03 0x00 0x2c 0x04 0x95 0x07 0x96 0x07 |..,.....
                               0xfa 0x01 0x8d 0x01                     |....
0xdb SAMPLES_FOR_AVG           0x82 0xff 0xff 0xff 0xff 0xff 0xff 0xff
0xdc READ_AVG_VIN              0x0795 = 42.577V
0xdd READ_AVG_VOUT             0x0794 = 42.817V
0xde READ_AVG_IIN              0x042b
0xdf READ_AVG_PIN              0x01f9
0xe0 BLACK_BOX_READ            0x02 0x00 0x0a 0x00 0x97 0x07 0x98 0x07 |........
                               0x04 0x00 0x6a 0x01                     |..j.
0xe1 READ_DIAGNOSTIC_WORD      0x0003
0xe2 AVG_BLOCK_READ            0x03 0x00 0x2b 0x04 0x94 0x07 0x95 0x07 |..+.....
                               0xf9 0x01 0x89 0x01                     |....
```

ericaasen, 2025-05-20:
i'm not sure why `SAMPLES_FOR_AVG` is shown as an 8-byte value since it should only be 1

[image]

mkeeter, 2025-06-18:
Averaging was added in #2100