oxidecomputer/hubris #2078: determining host OS ROM contents via management network

jclulow, 2025-05-21:
When using `humility qspi -D` to update the host OS ROM image, the differential write facility (I believe) first essentially hashes the current flash contents and returns hashes (at some block granularity) over the contents.  This is deemed to be quicker than spending 5+ minutes writing the entire ROM over USB.  It has the knock-on effect that we can relatively cheaply tell when the correct host OS ROM is _already present_ without eating a write cycle on the flash chip.

Over the management network, because there is so much more bandwidth available, I believe we just post a host OS ROM image update and the SP writes it unconditionally, which is still a lot faster than even most differential updates via USB/SWD.

For automation, it would be good to be able to just ask the SP, over the management network, what ROM image is present there already.  My current use case is essentially a CI/CD system, where we may want to use one of a handful of different images depending on what the job requires.  At the start of the job, we want to confirm which ROM is loaded in either BSU 0 or BSU 1, and only load a new ROM image if that image is not present in either BSU.  Otherwise, we may end up writing the ROM rather frequently; e.g., if a job only takes a minute to run, we could get through quite a lot of write cycles in a day.

I think it would probably be good, if it's cheap, to be able to lift some kind of SHA hash value out, that we could then compare with the `rom` file we have for the image itself.  I'm not wedded to this, though: if reading the whole ROM image back out over the network is cheaper or faster or better in some way, I'm happy to do that and just compare the contents myself!

jclulow, 2025-05-21:
Also, I'm not super across the specifics but presumably if we did do a hash thing, it would have to account for the fact that we store the persistent boot settings in the zero page of the BSU 0 flash, and maybe skip or treat that part of the image as zeroes?

labbott, 2025-05-22:
The existing `humility qspi -D` hashes via sectors (`64*1024`) and then compares the whole file. Some of that choice is probably due to `hiffy` limitations. Worst case we could use the same algorithm but let me experiment.

jgallagher, 2025-05-22:
We haven't started the reconfigurator work to update host OS images yet (soon!), but I believe it will want this also. I think the most natural thing for it would be if we could ask for the `ArtifactHash` (in the TUF repo sense) of the phase 1 image, which I believe is the sha256 of `image/rom` within the host OS artifact.

> if we did do a hash thing, it would have to account for the fact that we store the persistent boot settings in the zero page of the BSU 0 flash, and maybe skip or treat that part of the image as zeroes?

+1 to this, with the nit that I think we need to treat that part of the image as all `0xff`. (control-plane-agent [checks for this when receiving a new image](https://github.com/oxidecomputer/hubris/blob/4d468e607dd1c7436985d7dc236544b9e58e0fee/task/control-plane-agent/src/update/host_flash.rs#L317-L332).)

labbott, 2025-05-23:
I played around with this and I think I have something that works on gimlet

```
$ FAUX_MGS=~/faux-mgs-hf time pilot -r berlin sp exec -e 'host-flash-hash' BRM42220011
May 23 13:23:46.740 INFO creating SP handle on interface berlin_sw1tp0, component: faux-mgs
May 23 13:23:46.803 INFO initial discovery complete, addr: [fe80::aa40:25ff:fe04:209%23]:11111, interface: berlin_sw1tp0, socket: control-plane-agent, component: faux-mgs
[34, f8, c3, 78, fc, 96, a0, 46, b, ac, b1, e4, b3, 37, 5c, c8, 6b, 83, a5, f6, 43, 4e, 23, 86, 34, 5b, 14, 96, c3, 7e, 93, 2b]

real       14.4
user        0.0
sys         0.0
$ sha256sum /data/local/env/berlin/will-update-2025-05-19/os/image/rom 
34f8c378fc96a0460bacb1e4b3375cc86b83a5f6434e2386345b1496c37e932b  /data/local/env/berlin/will-update-2025-05-19/os/image/rom
```

This work on the active slot. Because of the way the driver works, you need to change the active slot to get the hash of the other slot.

Need to do some more clean up and also work on cosmo.

labbott, 2025-05-23:
> > I played around with this and I think I have something that works on gimlet
> 
> Awesome!
> 
> > real       14.4
> 
> Is most of this time spent hashing the contents? Is control-plane-agent blocked waiting for another task while it's going on?
> 

I haven't profiled it. This was the obvious approach which leveraged the existing hash approach. The control plane will be blocked while this is happening.

> > Because of the way the driver works, you need to change the active slot to get the hash of the other slot.
> 
> This seems like a bit of a step backwards compared to [#2043](https://github.com/oxidecomputer/hubris/pull/2043) (where we can write either slot without having to change which is active).
> 

The active slot was me being lazy and not wanting to add it as an argument to mgs. Thanks for the reminder that just going with the active slot is the more confusing behavior.

> This may not make sense, but I wonder if there's some way we could address both of these. Maybe compute the hash asynchronously and cache the results?

We might be able to get better performance by tuning how much we're reading (Pretty sure saying "performance issue" will also summon several colleagues to give ideas too). If not we could try "start hash" and "get hash" commands but that will significantly complicate the hashing logic.

labbott, 2025-06-26:
This should be available now